Yurik72/ESPHap #99: wolfSSL_3_13_0.rar cannot open sometime

paultang, 2021-07-15:
download the rar  file. but it cannot be open with a lot unzip rar application.

paultang, 2021-07-15:
wolfssl.rar  has the same issue.  can you download it and try .thanks.

paultang, 2021-07-15:
To simplify installation, ESPHap comes with pre-configured versions of wolfSSL which can be used out of the box. Extract either wolfSSL_3_13_0.rar or wolfssl.rar into an Arduino library folder named "wolfssl"..
---------------
the two rar file cannot be extracted. please check it. thanks

HSkul, 2022-12-23:
While it doesn't work with unrar nor Ark (Ubuntu), WinRAR running under Wine (installed as Snap on Ubuntu) has no problem opening the rar files.

  Hope it helps.

  H