Yurik72/ESPHap #100: AI THINKER Error during compilation

thierryspath, 2021-07-19:
Hi ,``
somebody available to help me ?

Regards

Thierry
Arduino: 1.8.15 Hourly Build 2021/05/31 10:34 (Mac OS X), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

/Users/thierryspath/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libwpa_supplicant.a(base64.o): In function `base64_encode':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/wpa2/utils/base64.c:31: multiple definition of `base64_encode'
libraries/ESPHap/esphap_base64.c.o:/Users/thierryspath/Documents/Arduino/libraries/ESPHap/esphap_base64.c:39: first defined here
collect2: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
 Used: /Users/thierryspath/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /private/var/folders/qn/lhgxj_5j31lck78h9xfqr3600000gn/T/AppTranslocation/853BB767-B2A9-4945-80A1-F340E9ED31F5/d/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board AI Thinker ESP32-CAM.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Yurik72, 2021-07-19:
See #79, #83 be sure that you uses latest version/commits

thierryspath, 2021-07-19:
> See #79, #83 be sure that you uses latest version/commits

Thank you for the answer 

[image]

last version ESPHAP & WolfSSL libraries
New test and same message

Arduino: 1.8.15 Hourly Build 2021/05/31 10:34 (Mac OS X), Board: "AI Thinker ESP32-CAM, 240MHz (WiFi/BT), QIO, 80MHz"

/Users/thierryspath/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libwpa_supplicant.a(base64.o): In function `base64_encode':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/wpa2/utils/base64.c:31: multiple definition of `base64_encode'
libraries/ESPHap-1.1/esphap_base64.c.o:/Users/thierryspath/Documents/Arduino/libraries/ESPHap-1.1/esphap_base64.c:39: first defined here
Multiple libraries were found for "WiFi.h"
collect2: error: ld returned 1 exit status
 Used: /Users/thierryspath/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /private/var/folders/qn/lhgxj_5j31lck78h9xfqr3600000gn/T/AppTranslocation/A5BE11CD-148A-4FB1-A12D-F818A310B440/d/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board AI Thinker ESP32-CAM.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Yurik72, 2021-07-20:
Ok, you are right. I will prepare new version updatable from Arduino IDE, those changes commited but unavailable for installation from Arduino IDE. Now you can simple get latest commit and copy to the your  Arduino Library folder (replace existing, which installed by Arduino IDE)

thierryspath, 2021-07-20:
> Ok, you are right. I will prepare new version updatable from Arduino IDE, those changes commited but unavailable for installation from Arduino IDE. Now you can simple get latest commit and copy to the your Arduino Library folder (replace existing, which installed by Arduino IDE)

Hi , i'm a newbie so i will wait the new version .
Thank you in advance for this nice job
Regards

Thierry

loenrrafael, 2022-06-02:
> Just get latest version directly from github and try to compile, should be fixed

I'm using release v1.1.