Yurik72/ESPHap #102: Clear pairing information

a11009, 2021-07-31:
I'm sorry to bother you . My C language is poor , so i can not clear pairing information. I would appreciate you if  you can help me package a function or teach me some details in order to remove the pairing information ?  Thank you very much .

init_hap_storageReaded bytes ->1409
>>> HomeKit: init_storage_ex size 0x581
>>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
>>> Home Integration: hap_services added chararacteristic  1: 00000025-0000-1000-8000-0026BB765291 
>>> Home Integration: hap_init_homekit_server library version v1.1
>>> Home Integration: homekit_is_paired 1
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 29:CA:D3:95:26:75
>>> HomeKit: Found admin pairing with 5F24A199-146E-4BEA-9908-E8DC95FAA940, disabling pair setup
>>> HomeKit: Configuring mDNS

a11009, 2021-07-31:
When i try to add SPIFFS.remove("/pair.dat"); to the code，the problem seems to have been solved。

Yurik72, 2021-08-02:
There are a few possibility to remove existing pairings:

1. When device is online (connected to Wifi) pairing can be removed from Apple Home App. Just remove deivice. But if device is not online pairing will no be removed on device and you need next  steps..
2.  You can remove pairing.dat from the web interface , see readme section
3. You can remove pairing.dat from the code (you exactly mentioned about that) and as well some logic can be implemented. Like long press on the some button can trigger removing pairing.dat

closed 2021-08-02 by Yurik72

a11009, 2021-08-18:
Okey,I have already solved this problem,thanks your reply.