Yurik72/ESPHap #103: Max number of switches on ESP32 (M5Stack core) ??

chrwh, 2021-08-19:
Hi,

Not sure this is a bug, it's probaly just me ;-)

When I use my below code with 6 switches, it all works as it should, but when I try to ad 1 or 2 more, it goes into a reboot loop. The 2 final switches are commented out in the code below.

Any pointers would be much appriciated.

`// BASIC CONFIGURATION 

//#define ENABLE_WIFI_MANAGER  // if we want to have built-in wifi configuration
                             // Otherwise direct connect ssid and pwd will be used
                             // for Wifi manager need extra library //https://github.com/tzapu/WiFiManager

#define ENABLE_WEB_SERVER    //if we want to have built in web server /site
#define ENABLE_OTA  //if Over the air update need  , ENABLE_WEB_SERVER must be defined first
#include <Arduino.h>
#include <M5Stack.h>
//RMT Test
#include "driver/rmt.h"
//#include <driver/rmt.h>
//#include <driver/periph_ctrl.h>

//Port pin 16 brugt til IR ud

#define RMT_TX_CHANNEL RMT_CHANNEL_0
#define RMT_TX_GPIO    2 // IR transmit output pin

#define PULSE_ON_T1  200
#define PULSE_OFF_T1 2925    // 3125 // 2925
#define PULSE_OFF_T2 6050    // 6250 // 6050
#define PULSE_OFF_T3 9175    // 9375 // 9175
#define PULSE_OFF_T4 12300   // 12500 // 12300
#define PULSE_OFF_T5 15425   // 15625 // 15425

#define BEO4_CMD_SIZE        22                     // sizeof(beo4_cmds) / sizeof(beo4_cmds[0])
#define ALL_OFF_OFFSET       (BEO4_CMD_SIZE * 0)
#define AUDIO_GO_OFFSET      (BEO4_CMD_SIZE * 1)
#define AUDIO_LIST_OFFSET    (BEO4_CMD_SIZE * 2)
#define AUDIO_VUP_OFFSET     (BEO4_CMD_SIZE * 3)
#define AUDIO_VDOWN_OFFSET   (BEO4_CMD_SIZE * 4)
#define AUDIO_CD_SELECT      (BEO4_CMD_SIZE * 5)
#define AUDIO_AUX_SELECT     (BEO4_CMD_SIZE * 6)
#define AUDIO_MUTE           (BEO4_CMD_SIZE * 7)

static bool firstIn_CD = 0; // Only change state if it's the firste time we are in the sub
static bool firstIn_AUX = 0; // Only change state if it's the firste time we are in the sub
static bool firstIn_OFF = 0; // Only change state if it's the firste time we are in the sub
static bool firstIn_VolUp = 0; // Only change state if it's the firste time we are in the sub
static bool firstIn_VolDown = 0; // Only change state if it's the firste time we are in the sub
static bool firstIn_Mute = 0; // Only change state if it's the firste time we are in the sub

// Remote Control Commands Using 1MHz clock
static const rmt_item32_t beo4_cmds[] = {
  // All Off (0x0F0C)
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T1, 0 }}}, // Start Bit 1 - AGC
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T1, 0 }}}, // Start Bit 2 - AGC
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T5, 0 }}}, // Start Bit 3 - Start
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T1, 0 }}}, // Start Bit 4 - Link Bit
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}}, // Data Start
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T3, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T1, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
  {{{ PULSE_ON_T1, 1, PULSE_OFF_T2, 0 }}},
[... 856 lines cut ...]
//          ch_Next->value.bool_value=val;
//          homekit_characteristic_notify(ch_Next,ch_Next->value);
//        }
//    }
//  }
//}
//void switch7_callback(homekit_characteristic_t *ch_Next, homekit_value_t value, void *context) {
//    Serial.println("switch7_callback");
// set_switch7(ch_Next->value.bool_value);
//}
//// Switch Prev end

#ifdef ENABLE_WIFI_MANAGER
void startwifimanager() {
  WiFiManager wifiManager;

wifiManager.setConfigPortalTimeout(120);
wifiManager.setConnectTimeout(60);

  if (!wifiManager.autoConnect(HOSTNAME, NULL)) {
      ESP.restart();
      delay(1000);
   }
}
#endif`

Yurik72, 2021-08-20:
Hi could you pleaase check if you are using latest version
As well please check homeintegration.c  
Looks you are on above off allowed range . So what is define for  MAX_HAP_SERVICES
Should be 
#define ESPHAP_VERSION "v1.1"
#ifdef ESP8266
#define MAX_HAP_SERVICES 8
#define MAX_HAP_ACCESSORIES 8
#else // ESP32.
#define MAX_HAP_SERVICES 45
#define MAX_HAP_ACCESSORIES 45
#endif

As weel please comment any logic in your callbacks ( set_switch6)
I see you are usign RMT, but not able briefly analyze if this not raise an error

chrwh, 2021-08-20:
Hi,
Thanks for the answer, I had tried the latest version thru Arduino, but that did not work. Downloaded the latest from here and installed it manually, and now, everythings is working as it should.

I would like you to know, that I really LIKE this software. I use it several places in my house :-) so THANKS for making this

Best regards
Christian

closed 2021-08-20 by chrwh