Yurik72/ESPHap #104: Initial load errors in webpage

chrwh, 2021-08-28:
Hi,
When ever I load the webpage, I get sporadic feedback for the stages of the the sliders. They does not get the correct response. In the picture all switches are off, when the page loaded.
If I then turn all the sliders to off, and then start to use them, they works correctly, until the next time you reload the page.

Do you have any pointers where I should start to look for my mistakes first?

[image]

chrwh, 2021-08-28:
Hmm,
It works with out any issues in Firefox, but with Edge and Chrome it has the issues above.

chrwh, 2021-08-29:
Found it, it's an issue in Webserver.cpp in ESP32 V.1.0.4, it has been fixed in V.1.0.6
// Fix for issue with Chrome based browsers: https://github.com/espressif/arduino-esp32/issues/3652
//           if (_currentClient.connected()) {
//             _currentStatus = HC_WAIT_CLOSE;
//             _statusChange = millis();
//             keepCurrentClient = true;
//           }

Yurik72, 2021-08-29:
Hi I know a root cause, by default esp is able to process one request at one time, means when is first request no finished second cold not be processed , there are another implementation exists, but for our case I w’d suggest and will do changes in jawascript to prevent second request when first is no finished

chrwh, 2021-08-29:
OK, that sounds good, let me know if I should make tests for you :-)

phil31, 2021-09-22:
just update to V3.3.2 and it seem not fixed for me !
my web site do a GET request to the ESP32 every 2 seconds

all requests are 200 OK then every 10 or 15 seconds, the server seem hangs .. 
request are (pending) for some seconds, maybe 10, then served or timeout for certain

using Chrome, Brave or Edge
[image]

chrwh, 2021-12-26:
Hi,
I stil lhave the same issues aswell. I would alsp really like a solution to this :-)
Best regards,
Christian