Yurik72/ESPHap #105: Unable to Add Accessory

alisadeghy, 2021-09-05:
Hi,
I'm using a Generic ESP8266 to make accessories. I could pair accessories a couple of times after so many tries but almost always while pairing the accessories it take too long and in the end says Unable to Add Accessory.
I'm using ESP8266 library version 3.0.2 (Also tries 2.7.0), IDE 1.8.15 iOS 14.7.1 on iPhone 7 Plus and latest ESPHap library.
What should I do? Please help me with that

Does ESP32 have such this problem too? 

Thanks

Yurik72, 2021-09-05:
Esp32 is much better, but for the esp8266, since latest versions there are not any problem on our huge community, check if you are using latest version and attach logs

alisadeghy, 2021-09-26:
The problem is about the board, maybe it's shorted or something. see more #83

closed 2021-09-26 by alisadeghy