Yurik72/ESPHap #106: ESPHAPCamera can't connect to Homekit - Wrong code?

chrwh, 2021-09-08:
Hi,
Just got a M5 Timer Camera F, it's up and running, but I can't add it to HomeKit, when trying to pair, first time it says "Wrong code"? If trying again, it says "Failed with OSStatus code -6700 (kUnknownErr) Any pointers

ESPHap V.1.1
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Christian\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.9\libraries\WiFi
 Not used: C:\Users\Christian\Documents\Arduino\libraries\WiFiNINA
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Using library WiFi at version 1.0 in folder: C:\Users\Christian\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.9\libraries\WiFi 
Using library ESPHap at version 1.1 in folder: C:\Users\Christian\Documents\Arduino\libraries\ESPHap 
Using library EEPROM at version 1.0.3 in folder: C:\Users\Christian\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.9\libraries\EEPROM 
Using library Ticker at version 1.1 in folder: C:\Users\Christian\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.9\libraries\Ticker 
Using library WiFiClientSecure at version 1.0 in folder: C:\Users\Christian\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.9\libraries\WiFiClientSecure 
Using library BLE at version 1.0.1 in folder: C:\Users\Christian\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.9\libraries\BLE 
Using library wolfSSL in folder: C:\Users\Christian\Documents\Arduino\libraries\wolfSSL (legacy)
"C:\\Users\\Christian\\AppData\\Local\\Arduino15\\packages\\m5stack\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\CHRIST~1\\AppData\\Local\\Temp\\arduino_build_792702/ESPHAPCamera.ino.elf"

Serial log:
11:14:36.495 -> WiFi connected
11:14:36.495 -> E (5677) fr_flash: Not found
11:14:36.495 -> Starting web server on port: '80'
11:14:36.495 -> Starting stream server on port: '81'
11:14:36.495 -> GPIO setup
11:14:36.495 -> setup_servo_pin
11:14:36.541 -> setup_internal_loop 
11:14:36.541 -> setup_internal_loop  entered 
11:14:36.541 -> Camera started -> done
11:14:36.541 -> Camera Ready! Use 'http://192.168.1.239' to connect
11:14:36.541 -> init_hap_storageStorage size 289 
11:14:36.541 -> >>> HomeKit: init_storage_ex size 0x121
11:14:36.541 -> >>> Home Integration: hap_add_motion_service
11:14:36.541 -> >>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
11:14:36.541 -> >>> Home Integration: hap_services added chararacteristic  1: 00000022-0000-1000-8000-0026BB765291 
11:14:36.541 -> >>> Home Integration: hap_init_homekit_server library version v1.1
11:14:36.588 -> >>> Home Integration: homekit_is_paired 0
11:14:36.588 -> >>> HomeKit: Starting server
11:14:36.588 -> >>> HomeKit: Formatting flash at 0x200000
11:14:36.588 -> unable to write erprom>>> HomeKit: Generated new accessory ID: 44:40:CC:BF:F2:F2
11:14:36.588 -> unable to write erprom>>> HomeKit: Generated new accessory key
11:14:36.588 -> unable to write erprom>>> HomeKit: Configuring mDNS
11:14:58.253 -> >>> HomeKit: Got new client connection: 61
11:14:58.253 -> >>> HomeKit: [Client 61] Pair Setup 
11:14:58.253 -> >>> HomeKit: [Client 61] Pair Setup Step 1/3
11:14:58.253 -> >>> HomeKit: [Client 61] new pairing content
11:14:58.253 -> >>> HomeKit: started crypto
11:14:58.253 -> >>> HomeKit: Free heap: 3301744
11:15:00.354 -> >>> HomeKit: Calculating public key len=384
11:15:01.429 -> >>> HomeKit: Calculating public key result 0
11:15:01.429 -> >>> HomeKit: [Client 61] send_tlv_response
11:15:01.429 -> >>> HomeKit: [Client 61] send_tlv_response done
11:15:01.569 -> >>> HomeKit: [Client 61] Pair Setup 
11:15:01.569 -> >>> HomeKit: [Client 61] Pair Setup Step 2/3
11:15:04.371 -> !!! HomeKit: [Client 61] Failed to compute SRP shared secret (code -2)
11:15:04.371 -> >>> HomeKit: [Client 61] Pair Setup 
11:15:04.371 -> >>> HomeKit: [Client 61] Pair Setup Step 1/3
11:15:04.371 -> >>> HomeKit: Failed to set SRP password (code -173)
11:15:04.417 -> >>> HomeKit: Calculating public key len=384
11:15:05.445 -> >>> HomeKit: Calculating public key result 0
11:15:05.491 -> >>> HomeKit: [Client 61] send_tlv_response
11:15:05.491 -> >>> HomeKit: [Client 61] send_tlv_response done
11:15:05.630 -> >>> HomeKit: [Client 61] Closing client connection
11:26:22.894 -> Motion detected 639:
11:26:22.894 -> Notify Motion
11:26:22.894 -> 63944
11:27:14.611 -> JPG: 66138B 99ms
11:34:16.358 -> >>> HomeKit: Got new client connection: 61
11:34:16.358 -> >>> HomeKit: [Client 61] Pair Setup 
11:34:16.358 -> >>> HomeKit: [Client 61] Pair Setup Step 1/3
11:34:16.358 -> >>> HomeKit: [Client 61] new pairing content
11:34:16.405 -> >>> HomeKit: started crypto
11:34:16.405 -> >>> HomeKit: Free heap: 3449700
11:34:16.637 -> >>> HomeKit: Calculating public key len=384
11:34:16.729 -> >>> HomeKit: Calculating public key result -2
11:34:16.729 -> !!! HomeKit: [Client 61] Failed to dump SPR public key (code -2)
11:34:16.869 -> >>> HomeKit: [Client 61] Closing client connection
11:35:15.116 -> >>> HomeKit: Got new client connection: 63
11:35:15.116 -> >>> HomeKit: [Client 63] Pair Setup 
11:35:15.116 -> >>> HomeKit: [Client 63] Pair Setup Step 1/3
11:35:15.116 -> >>> HomeKit: [Client 63] new pairing content
11:35:15.116 -> >>> HomeKit: started crypto
11:35:15.116 -> >>> HomeKit: Free heap: 3449708
11:35:15.256 -> >>> HomeKit: Calculating public key len=384
11:35:15.439 -> >>> HomeKit: Calculating public key result -2
11:35:15.439 -> !!! HomeKit: [Client 63] Failed to dump SPR public key (code -2)
11:35:15.579 -> >>> HomeKit: [Client 63] Closing client connection

chrwh, 2021-09-08:
Hmm, now I got i paired, but it only shows up as a motion sensor, and no camera ?

chrwh, 2021-09-08:
Tried again, and can't pair once again

Yurik72, 2021-09-09:
> Hmm, now I got i paired, but it only shows up as a motion sensor, and no camera ?

At least for this question answer is simple. It designed to use as Motion Sensor. No camera video stream implemented

Yurik72, 2021-09-09:
> Tried again, and can't pair once again

Potentially i see a problem. Camera is started and triyng to detect motion. It should not happen during the pairing process.
I will check and fix. Any how try to pair without any motion. Let's say close camera by any paper or put leans on the table

chrwh, 2021-09-09:
Too sad to hear that, will video be implemented soon or at all? as I see it, it's part of the HomeKit capabilities.

It will also only do motion detection 1 or 2 times, before it dies, and it looses connection to the homekit. Also I don't see how to reset the motion detection in the homekit? (on iPad)

Yurik72, 2021-09-09:
Good  evening, yes this is my plan to finish full capabilities of HomeKit including video streaming, but currently it’s only motion detection :( which is not trivial.) please  Explain what is the problem of resetting.  If any motion detected ( images are equals)  device should resetting

chrwh, 2021-09-09:
I know that none of the work you do here is trivial - and you should know that I REALLY appreciate all the hard work you do here 🙂👍🏻

When a motion has been detected, it sends a mail and notify my iPad. But the little red mark on the iPad icon never disappears again, and after a while it looses connection to the HomeKit

Best regards
 
Christian

chrwh, 2021-09-09:
Here is the log startin from new with blinded lense, and when paired to time looking at the camera, and blinding it again, iPad never get's the notice, and then looses connection to the homekit

22:20:15.325 -> WiFi connected
22:20:15.325 -> E (4850) fr_flash: Not found
22:20:15.371 -> Starting web server on port: '80'
22:20:15.371 -> Starting stream server on port: '81'
22:20:15.371 -> GPIO setup
22:20:15.371 -> setup_servo_pin
22:20:15.371 -> setup_internal_loop 
22:20:15.371 -> setup_internal_loop  entered 
22:20:15.371 -> Camera started -> done
22:20:15.371 -> Camera Ready! Use 'http://192.168.1.239' to connect
22:20:15.371 -> init_hap_storageStorage size 289 
22:20:15.371 -> >>> HomeKit: init_storage_ex size 0x121
22:20:15.371 -> >>> Home Integration: hap_add_motion_service
22:20:15.371 -> >>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
22:20:15.371 -> >>> Home Integration: hap_services added chararacteristic  1: 00000022-0000-1000-8000-0026BB765291 
22:20:15.419 -> >>> Home Integration: hap_init_homekit_server library version v1.1
22:20:15.419 -> >>> Home Integration: homekit_is_paired 0
22:20:15.419 -> >>> HomeKit: Starting server
22:20:15.419 -> >>> HomeKit: Using existing accessory ID: 37:28:DE:C2:E5:1E
22:20:15.419 -> >>> HomeKit: Configuring mDNS
22:20:23.488 -> >>> HomeKit: Got new client connection: 61
22:20:23.488 -> >>> HomeKit: [Client 61] Pair Setup 
22:20:23.488 -> >>> HomeKit: [Client 61] Pair Setup Step 1/3
22:20:23.535 -> >>> HomeKit: [Client 61] new pairing content
22:20:23.535 -> >>> HomeKit: started crypto
22:20:23.535 -> >>> HomeKit: Free heap: 3304200
22:20:25.589 -> >>> HomeKit: Calculating public key len=384
22:20:26.616 -> >>> HomeKit: Calculating public key result 0
22:20:26.616 -> >>> HomeKit: [Client 61] send_tlv_response
22:20:26.616 -> >>> HomeKit: [Client 61] send_tlv_response done
22:20:26.847 -> >>> HomeKit: [Client 61] Pair Setup 
22:20:26.847 -> >>> HomeKit: [Client 61] Pair Setup Step 2/3
22:20:30.345 -> >>> HomeKit: [Client 61] Pair Setup 
22:20:30.345 -> >>> HomeKit: [Client 61] Pair Setup Step 3/3
22:20:30.391 -> unable to write erprom>>> HomeKit: Added pairing with 9072A330-BDA9-42C6-B85B-41DA77A21220
22:20:30.437 -> >>> HomeKit: Configuring mDNS
22:20:30.437 -> >>> HomeKit: [Client 61] Successfully paired
22:20:30.578 -> >>> HomeKit: [Client 61] Closing client connection
22:20:30.673 -> >>> HomeKit: Got new client connection: 62
22:20:30.673 -> >>> HomeKit: [Client 62] Pair Verify Step 1/2
22:20:30.858 -> >>> HomeKit: [Client 62] Pair Verify Step 2/2
22:20:30.858 -> >>> HomeKit: [Client 62] Found pairing with 9072A330-BDA9-42C6-B85B-41DA77A21220
22:20:30.905 -> >>> HomeKit: [Client 62] Verification successful, secure session established
22:20:30.997 -> >>> HomeKit: [Client 62] Get Accessories
22:20:31.091 -> >>> HomeKit: [Client 62] Update Characteristics
22:20:31.091 -> >>> HomeKit: [Client 62] Update Characteristics
22:20:31.091 -> >>> HomeKit: [Client 62] Update Characteristics
22:20:33.007 -> >>> HomeKit: Got new client connection: 63
22:20:33.007 -> >>> HomeKit: [Client 63] Pair Verify Step 1/2
22:20:33.191 -> >>> HomeKit: [Client 63] Pair Verify Step 2/2
22:20:33.191 -> >>> HomeKit: [Client 63] Found pairing with 9072A330-BDA9-42C6-B85B-41DA77A21220
22:20:33.238 -> >>> HomeKit: [Client 63] Verification successful, secure session established
22:20:33.238 -> >>> HomeKit: [Client 63] Get Accessories
22:23:38.003 -> Motion detected 1193:
22:23:38.003 -> Notify Motion
22:23:38.003 -> 65616
22:23:40.796 -> 89160
22:23:40.842 -> Connected
22:23:43.553 -> Sending status: 
22:23:43.553 -> 1
22:23:43.553 -> 0
22:23:43.553 -> Message sent!
22:23:43.553 -> >>> Home Integration: hap_init_homekit_server library version v1.1
22:23:43.553 -> >>> Home Integration: homekit_is_paired 1
22:23:43.553 -> >>> HomeKit: Starting server
22:23:43.553 -> E (213048) httpd: httpd_server_init: error in bind (112)
22:23:43.553 -> >>> HomeKit: Using existing accessory ID: 37:28:DE:C2:E5:1E
22:23:43.553 -> >>> HomeKit: Found admin pairing with 9072A330-BDA9-42C6-B85B-41DA77A21220, disabling pair setup
22:23:43.599 -> >>> HomeKit: Configuring mDNS
22:24:42.608 -> Motion detected 1530:
22:24:42.608 -> 72848

Yurik72, 2021-09-10:
Ok, seems that is bug if motion is not detected  a while ESP should update Apple about that . I will check