Yurik72/ESPHap #107: How can I make a LED Light with Color Temperature (No RGB)?

alisadeghy, 2021-09-26:
Hi, I want to make a LED light with color temperature control.
It's made using 2 sets of cool and warm LEDs, how can I implement that in accessory?
I don't want it to show RGB tab in Home app, just temperature tab.

Thanks

Yurik72, 2021-09-27:
I think you means some kind of adaptive lighting, 
Pleased have a look #95 , it seems wil answers to your question

a11009, 2022-07-23:
> No I didn't mean that. I mean bicolor lights which we only have control over color temperature. It's usually made with 2 sets of cool and warm LEDs.

Did you solve it? I'm also working on a project of mixing color temperature between warm and cold lighting recently.

a11009, 2022-07-23:
> I think you means some kind of adaptive lighting, Pleased have a look #95 , it seems wil answers to your question

What he meant was adjusting the color temperature using only cool and warm lights, not using RGB. I also want to do something in this area, but I have not fully understood your project due to the level, I wonder if you can give me some pointers?

bartosz-777, 2024-02-05:
It works in HomeSpan, I know what you mean,commonly called "cct" light, it uses "ColorTemperature" HomeKit characteristic instead of hsv, the value of it isn't in Kelvins, to convert divide 1,000,000 by it (K = 1000000/value)