Yurik72/ESPHap #114: Getting configured accessory name from HomeKit

fluetke, 2021-12-18:
Is there a way to get the configured accessory name in HomeKit using ESPHap?
As in the identifier configured on the iPhone?

I'm currently trying to display the iOS-Name of an accessory on the integrated Webserver of the ESP, so that I can identify which configuration I'm working on. Think of having 5 lights in the ceiling and having no idea which one you're currently setting up using the webinterface.

Is there some kind of HAP_GET_ACCESSORY_NAME or something like that?

Yurik72, 2021-12-22:
Hi, could you please explain where you expected to look accessory name. On the browser ?
In Apple home kit application you have chanse:
1. To rename device
2.  During the setup (see example ESPHapAdvanceLed)  you ca give any name for accessory and services. just change string values on the code below

```
   hap_initbase_accessory_service(HOSTNAME,"Yurik72","0","EspHapLed","1.0");

   //we will add only one light bulb service and keep pointer for nest using
    hapservice= hap_add_lightbulb_service("Led",led_callback,(void*)&led_gpio);
```

closed 2021-12-22 by Yurik72

fluetke, 2021-12-22:
Hi, 

basically I'm looking for a way to retrieve the device-name _after_ it was paired and changed in the HomeKit/Apple Home app.
Let's say I have two lights A and B, both are using the same firmware with ESPHap and do a 

```
hap_initbase_accessory_service("lightbulb", "Author", "0", "EspHapLed", "1.0");
service = hap_add_lightbulb_service("led", callback, (void*) &led_gpio);

```
They will show up in the HomeKit app on the iPhone as 
A- EspHapLed-R4ND
B- EspHapLed-ND0M

After pairing I can assign a new name for these Devices in the HomeKit app like for example
A- WindowLight
B- CeilingLight

What I'm trying to is to read these changed names back into the firmware, so that the "given name" can be displayed on the device websites of bulbs A and B, where more settings can be configured than in the HomeKit app on iPhone alone.

So that, when opening the device website (for example to check the power-usage of a device), I can display the same name to the user as is configured in the iPhone Home app, making it easier to identify the correct device compared to a default model-name with a random-device id.

Maybe an even better example for this use case would be 2 window coverings on differently sized windows, where you need to configure the start and stop points on the device website. I just want to display the new device names on the configuration website of a device.