Yurik72/ESPHap #116: Memory leak

ck0567, 2022-01-03:
Hi, i am using the function homekit_characteristic_notify() in a loop to inform apple HomeKit
about status changes. After pairing process the Heap memory is decreasing so that the esp32 crashes
after a while.
Any ideas ?
Thanks

Yurik72, 2022-01-03:
Hi, I have tested a lot for memory leak and seems they should not be.. Hovewer for clear answer I need your full   sketch .
Without that not,  any ideas... Maybe you have call this function a lot, until previous is not finished. So, if you are using homekit_characteristic_notify() in the loop try any kind of delay to call this function 
As well try to call this function if your values is really changed. For instance I have imlemented a macros
HAP_NOTIFY_CHANGES... by using that, it will compare previous value, before sending real request

P.S. be sure that you are using the latest version of library

ck0567, 2022-01-03:
Yes, you are right. I have implemented a timer to call the update function and it works
without decresing heap memory. Thank you so much.
Chris

Yurik72, 2022-01-03:
Good, seems we can close an issue as well known  feathure :)

closed 2022-01-03 by Yurik72