Yurik72/ESPHap #119: Accessories works on one phone but not responding on another

matrixall, 2022-04-11:
I have used basic ESPHAP led example on multiple ESP8266-12F (11 Accessories) which worked perfect for couple of days for me and my wife. However, for the past week. it is not responsive to me but works just fine for my wife which I invited here. is there a client limit to the accessories or something I'm not aware of.

Yurik72, 2022-04-12:
Hi for ESP8266 max connection are limited to 8
#define HOMEKIT_MAX_CLIENTS  8  (arduino_homekit_server.cpp line 42)
Be sure that you are using latest version of this library

matrixall, 2022-04-13:
I'm using the latest version 1.07, it is as default. However, I restarted my phone **IPhone X**, the problem went away for few hours and then comes back again with every phone restart. Thus, I decided to order a new IPhone and see if it presets.

Yurik72, 2022-04-13:
Ok, seems i can help you without ordering new iphone :)
1. You need to use latest version 1.1 (I'm not sure why library shows 1.07 as the latest) i will check. Anyhow you can simple get latest version from git. v1.1 has fixes some memory leak, which apperars after long usage of device
2. Please have a look #52, which is similar to your problem. You  should compile with option lwIP variant: v2 Higher Bandwidth , instead v2 Lower Memory. (Arduino menu)
regards

matrixall, 2022-04-23:
Hi, It has been over a week with the new IPhone (13) with no issues at all. Sorry for misinforming you before, the version I uses is 1.1 (don't know how I came up with 1.0.7). With regards to the second point. yes it is already implemented. Thanks for the support.

closed 2022-04-23 by matrixall