Yurik72/ESPHap #120: Door Lock Accessory

mmackh, 2022-04-11:
Has anyone managed to add a door lock service? Here's my attempt - but I do have an error. It also results in adding two devices.

`
  const char *serviceName = "InstaNFC Access Control";

  hap_initbase_accessory_service(serviceName,"company","0","IAP", "1.0");
   
  homekit_service_t *services[3];
  services[0] = hap_new_homekit_accessory_service(serviceName, "0");
  services[1] = hap_new_switch_service(serviceName,cb, NULL);
  services[2] = NULL;
  hap_add_accessory(homekit_accessory_category_door_lock, services);
  hap_add_service(services[1]);
  hapservice = services[1];

  lockCharacteristic = homekit_service_characteristic_by_type(hapservice, HOMEKIT_CHARACTERISTIC_ON);
  
  hap_init_homekit_server();`

mmackh, 2022-04-11:
Ok, I'm a little further along now. Only one device, but it's a working switch in this case.

```
  init_hap_storage(pair_file_name);
  hap_setbase_accessorytype(homekit_accessory_category_door_lock);
  hap_initbase_accessory_service("host","company", productID.c_str(),"AP", "1.0");
   
  const char *serviceName = "Access Control";
  hapservice = hap_add_switch_service(serviceName, lockCallback, NULL);
  lockCharacteristic = homekit_service_characteristic_by_type(hapservice, HOMEKIT_CHARACTERISTIC_ON);
  
  hap_init_homekit_server();
```

Yurik72, 2022-04-12:
Hi, Lock control is not implemented (ported) to my library.
if you want to see lock as accessory on the apple you have to implement the same function (like for switch) , but using
service : HOMEKIT_SERVICE_LOCK_MANAGEMENT
characteristic: 
/**
 Defines a number of additional settings, rules and information on a lock mechanism inside of a accessory.
 
 Required Characteristics:
 - LOCK_CONTROL_POINT
 - VERSION
 
 Optional Characteristics:
 - NAME
 - LOGS
 - AUDIO_FEEDBACK
 - LOCK_MANAGEMENT_AUTO_SECURITY_TIMEOUT
 - ADMINISTRATOR_ONLY_ACCESS
 - LOCK_LAST_KNOWN_ACTION
 - CURRENT_DOOR_STATE
 - MOTION_DETECTED
 */

mmackh, 2022-04-12:
Thank you for the fast response. I'm not yet sure on how to add a custom accessory type - everything is a macro so when starting off with a template like for the switch:

```
NEW_HOMEKIT_SERVICE(SWITCH, .primary = true,.characteristics=(homekit_characteristic_t*[]) {
	            NEW_HOMEKIT_CHARACTERISTIC(NAME, szname),
	            NEW_HOMEKIT_CHARACTERISTIC(
	                ON, true,
	                .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(
	                		cb, .context=context
	                ),
	            ),
	            NULL
	        });
```

I'm not able to get it to compile. Is there a function / method I'm missing somewhere to create custom services (incl. characteristics)?

Yurik72, 2022-04-13:
Ok, i will do that, but later , now do not have a time and energy (I'm from Kiev)
You can send me your file and changes i will try to compile
regards