Yurik72/ESPHap #123: ESP32 stucks on "HomeKit: Configuring mDNS"

EnesCinr, 2022-05-27:
I am trying to run the "EspHapLed" example with my ESP32 device. But when i upload the code the logs are here:

>>> Connecting to Superbox_WiFi_2.4GHz_98E3
.........
WiFi connected
IP address: 
192.168.8.125
init_hap_storageReaded bytes ->1409
>>> HomeKit: init_storage_ex size 0x581
>>> Home Integration: hap_services added chararacteristic  0: 00000023-0000-1000-8000-0026BB765291 
>>> Home Integration: hap_services added chararacteristic  1: 00000025-0000-1000-8000-0026BB765291 
>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: FC:0C:53:BD:2E:D1
>>> HomeKit: Configuring mDNS

And it stucks on there. I debugged the server.c code and i understood that, in this line [server.c:3380](https://github.com/Yurik72/ESPHap/blob/master/server.c#L3380) the 'triggered_nfds' variable is returning 0 value. So the for loop is going to infinity loop. 

I use ESPHap 1.1 version. With the esp8266 its completely working.

Yurik72, 2022-05-28:
Hi, I do not see any stucks on your logs. ESP is running and waiting for pairing.
Regarding server.c:3380 it's up to design , this is infinity loop (when server run) for incoming TCP connection. 
triggered_nfds will be not zero, when some connection come

EnesCinr, 2022-05-28:
> Hi, I do not see any stucks on your logs. ESP is running and waiting for pairing. Regarding server.c:3380 it's up to design , this is infinity loop (when server run) for incoming TCP connection. triggered_nfds will be not zero, when some connection come

thank you so much, i didnt think to look at homekit for add the device. I just thought this is a bug. Thank you again for this awesome library, have a good day.

closed 2022-05-28 by EnesCinr