Yurik72/ESPHap #127: Hi,

laszloferenczi, 2022-12-26:
H,

I would like to test Esphap_Switch examples. But after then configured wifi and try to pair device I get the following result.

>>> Home Integration: homekit_is_paired 0
>>> HomeKit: Update the CPU to run at 160MHz
>>> HomeKit: Free heap: 28616
>>> HomeKit: Starting server
>>> HomeKit: storage init 0
>>> HomeKit: Using existing accessory ID: 3F:39:4D:63:1F:D2
>>> HomeKit: find pairings
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 22272
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: Configuring mDNS
>>> HomeKit: Wifi Sleep type:2
>>> HomeKit: Set Wifi Sleep type :0
>>> HomeKit: MDNS.begin: host-3F39, IP: 192.168.1.18
>>> HomeKit: Init server over
Web server is NOT SET, waiting for pairing
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 192.168.1.18:5556, remote 192.168.1.10:56604
>>> HomeKit: [Client 1073705396] Got client content from the cache 0
>>> HomeKit: [Client 1073705396] new client accepted
>>> HomeKit: [Client 1073705396] Pair Setup Step 1/3
>>> HomeKit: [Client 1073705396] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073705396] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 20776
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success
>>> HomeKit: WiFiServer receives a new client (current 0, max 8)
>>> HomeKit: Got new client connection: local 192.168.1.18:5556, remote 192.168.1.10:56605
>>> HomeKit: [Client 1073703844] Got client content from the cache 0
>>> HomeKit: [Client 1073703844] new client accepted
>>> HomeKit: [Client 1073703844] Pair Setup Step 1/3
>>> HomeKit: [Client 1073703844] Connection: close received, client will be disconnected
>>> HomeKit: [Client 1073703844] Closing client connection
>>> HomeKit: Free saved_preinit_pairing_context
>>> HomeKit: [Client 0] Clear the pairing context
>>> HomeKit: [Client 0] client content free
>>> HomeKit: Preinit pairing context
>>> HomeKit: Using user-specified password: 111-11-111
>>> HomeKit: Free heap: 20704
>>> HomeKit: Calculating public key len=384
>>> HomeKit: Calculating public key result 0
>>> HomeKit: arduino_homekit_preinit success

And a pairing unsuccess.
Please help! 
Laci

Yurik72, 2022-12-27:
Hi, please have a look readme, you have to change compilation option.
Probably it will fix your issue
 **Note for ESP8266 boards select lwIP Variant :v2 High bandwidth. Device behaviours is more stable especially in pairing process**

laszloferenczi, 2022-12-28:
Thank you. As I see the issue is related to Homekit bridge mode. If i use separated accessories every thing is OK, when i try to use bridge not working. This time the separated accessory enought for me.

closed 2022-12-28 by laszloferenczi