Yurik72/ESPHap #139: All devices "No response" under iOS 16.6

eford321, 2023-08-02:
After 2.5 years, all of seven of my ESPHap devices have gone "No Response".   This seems to coincide with my hubs (HomePods) updating to iOS 16.6 under the new architecture.  Is anyone else seeing the same issue?

I attempted to remove and add devices in the home app and they timeout saying device is not reachable.

simkard69, 2023-09-13:
I had the same problem getting back from holidays.
On my side, it's my AppleTV 4K which acts as a hub.
Actually, none of my HomeKit accessories using this lib were working anymore.
I had to upgrade the HomeKit architecture on AppleTV's side and then, everything went back in business without the need of unpairing everything and pairing them again.

seamaster101, 2024-03-02:
Same problem here! All my devices stoped responding. Is there any fix coming or should we try ti find something else to use?